syssi/esphome-basen-bms #26: It works with a Roamer series two battery (200SMART2)

DrRob, 2023-11-26:
Hi,

Just to confirm that I've tried this with a Roamer series two battery (model: 200SMART2) and it appears to be working fine.  It only has four cells (from the BMS's point of view) so I just removed all but four from the config.  I also need to figure out which one of the four temperatures is the one that the Android app displays (it's showing 0, 25, 25, and 4, so it's not simply the case that the irrelevant ones are 0).

A small tangent - how straightforward would it be to take the information in the sources of this module and turn it into a Python script that I can run on a Raspberry Pi, do you think?  I like esphome but I'm not using it where I have this battery, and don't usually have a suitable wi-fi access point for the esp to connect to (I set one up just to test this).

Thanks for creating this.

Rob.

syssi, 2023-11-27:
Thanks for your feedback! I've added your battery to the list of supported devices. If you are familiar with python and BLE communication it should take a day to add the BMS support to the `mpp-solar` project for example.

DrRob, 2023-11-27:
Thanks for the pointer to mpp-solar, I hadn't come across that, will take a look.

virtualwebber, 2025-05-06:
@DrRob Any chance I could see what your conf looks like as trying to get this working with a S3 Roamer :)

DrRob, 2025-05-06:
@virtualwebber happy to, although I thought that the S3 Roamer had a different BMS...?

I'm running it on a esp32-c3-zero (https://thepihut.com/products/esp32-c3-mini-development-board) and also have a temp/humidity/pressure sensor attached.  It's sending updates to MQTT – I don't have HA running, I'm just using esphome for its MQTT integration.  It's connecting via wifi to a Pi acting as an wifi AP, which is only about 15cm away, and yet the wifi connection drops fairly frequently, and it often takes a long time to reappear – I'm not sure why it's so unreliable.  I will try a different microcontroller when I get time.

Here's the config with a few secrets redacted:

```substitutions:
  name: roamer
  device_description: "Roamer BMS monitor"
  external_components_source: github://syssi/esphome-basen-bms@main
  mac_address: A4:C1:38:XX:XX:XX   # REDACTED

esphome:
  name: ${name}
  friendly_name: "Roamer BMS monitor"
  comment: ${device_description}
  min_version: 2024.6.0
  platformio_options:
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf
    version: recommended

external_components:
  - source: ${external_components_source}
    refresh: 0s

#debug:
#  update_interval: 10s

wifi:
  networks:
    - ssid: !secret REDACTED
      password: !secret REDACTED
      manual_ip: 
        static_ip: 192.168.32.60
        gateway: 192.168.32.1
        subnet: 255.255.255.0
        dns1: 192.168.32.1
  use_address: 192.168.32.60
  fast_connect: True
  on_connect:
    - esp32_ble_tracker.start_scan:
        continuous: true
  on_disconnect:
    - esp32_ble_tracker.stop_scan:

#web_server:
#  port: 80

logger:
  level: INFO

mqtt:
  broker: 192.168.32.1
  port: 1883
  discovery: True
  skip_cert_cn_check: True
  certificate_authority: |
    -----BEGIN CERTIFICATE-----
    REDACTED
    -----END CERTIFICATE-----
  client_certificate: |
    -----BEGIN CERTIFICATE-----
[... 141 lines cut ...]
    charging_states:
      name: "${name} charging states"
    discharging_states:
      name: "${name} discharging states"
    charging_warnings:
      name: "${name} charging warnings"
    discharging_warnings:
      name: "${name} discharging warnings"
    manufacturing_date:
      name: "${name} manufacturing date"

  #- platform: debug
  #  reset_reason:
  #    name: "Reset Reason"

switch:
  #- platform: ble_client
  #  ble_client_id: client0
  #  name: "${name} enable bluetooth connection"

  - platform: basen_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"```

virtualwebber, 2025-05-06:
> [@virtualwebber](https://github.com/virtualwebber) happy to, although I thought that the S3 Roamer had a different BMS...?
> 
> I'm running it on a esp32-c3-zero (https://thepihut.com/products/esp32-c3-mini-development-board) and also have a temp/humidity/pressure sensor attached. It's sending updates to MQTT – I don't have HA running, I'm just using esphome for its MQTT integration. It's connecting via wifi to a Pi acting as an wifi AP, which is only about 15cm away, and yet the wifi connection drops fairly frequently, and it often takes a long time to reappear – I'm not sure why it's so unreliable. I will try a different microcontroller when I get time.
> 
> Here's the config with a few secrets redacted:
> 
> ```
>   name: roamer
>   device_description: "Roamer BMS monitor"
>   external_components_source: github://syssi/esphome-basen-bms@main
>   mac_address: A4:C1:38:XX:XX:XX   # REDACTED
> 
> esphome:
>   name: ${name}
>   friendly_name: "Roamer BMS monitor"
>   comment: ${device_description}
>   min_version: 2024.6.0
>   platformio_options:
>     board_build.f_flash: 40000000L
>     board_build.flash_mode: dio
>     board_build.flash_size: 4MB
> 
> esp32:
>   board: esp32-c3-devkitm-1
>   variant: esp32c3
>   framework:
>     type: esp-idf
>     version: recommended
> 
> external_components:
>   - source: ${external_components_source}
>     refresh: 0s
> 
> #debug:
> #  update_interval: 10s
> 
> wifi:
>   networks:
>     - ssid: !secret REDACTED
>       password: !secret REDACTED
>       manual_ip: 
>         static_ip: 192.168.32.60
>         gateway: 192.168.32.1
>         subnet: 255.255.255.0
>         dns1: 192.168.32.1
>   use_address: 192.168.32.60
>   fast_connect: True
>   on_connect:
>     - esp32_ble_tracker.start_scan:
>         continuous: true
>   on_disconnect:
>     - esp32_ble_tracker.stop_scan:
> 
> #web_server:
> #  port: 80
> 
> logger:
>   level: INFO
> 
> mqtt:
>   broker: 192.168.32.1
>   port: 1883
>   discovery: True
>   skip_cert_cn_check: True
>   certificate_authority: |
>     -----BEGIN CERTIFICATE-----
>     REDACTED
>     -----END CERTIFICATE-----
>   client_certificate: |
>     -----BEGIN CERTIFICATE-----
[... 144 lines cut ...]
>       name: "${name} discharging states"
>     charging_warnings:
>       name: "${name} charging warnings"
>     discharging_warnings:
>       name: "${name} discharging warnings"
>     manufacturing_date:
>       name: "${name} manufacturing date"
> 
>   #- platform: debug
>   #  reset_reason:
>   #    name: "Reset Reason"
> 
> switch:
>   #- platform: ble_client
>   #  ble_client_id: client0
>   #  name: "${name} enable bluetooth connection"
> 
>   - platform: basen_bms_ble
>     charging:
>       name: "${name} charging"
>     discharging:
>       name: "${name} discharging"```
> ```

@DrRob Thank you :). It was Roamer themselves that pointed me to this actually and said the S3 and S2 share the same BLE module. The S4 is totally different though as it uses CANbus and the Victron protocol (BMS-CAN).  Thanks again for sharing the configuration