syssi/esphome-basen-bms #29: Negative temperature readings show incorrectly

will-h, 2024-01-16:
Looks like the temperature is actually a signed int. The provided Roamer android app says my BMS temperature is -2°C, but esphome-basen-bms reports 254°C.

`[13:33:22][D][sensor:094]: 'basen-bms-ble temperature 1': Sending state 254.00000 °C with 0 decimals of accuracy`

syssi, 2024-01-16:
I've pushed a fix to a feature branch. Please update your YAML (temporary) from:

```
substitutions:
  name: basen-bms-ble
  device_description: "Monitor a Basen Battery Management System via BLE"
  external_components_source: github://syssi/esphome-basen-bms@main
```

to 

```
substitutions:
  name: basen-bms-ble
  device_description: "Monitor a Basen Battery Management System via BLE"
  external_components_source: github://syssi/esphome-basen-bms@fix-temperatures
```

Please flash your ESP another time and give it a try. Does this change fix the issue?

will-h, 2024-01-16:
That was quick, thank you! 

I have flashed the ESP but inconveniently the temperature is now above zero - I'll check this tomorrow.

closed 2024-01-16 by syssi

will-h, 2024-01-17:
To follow up - works fine and displays negative temperatures correctly.

syssi, 2024-01-18:
Perfect! I've merged the fix into `main` already. Please change your YAML configuration back to `@main`. Is your battery pack already mentioned as supported at the README or do you like to provide some details about your battery?

commit 3272169c (2024-01-16): Fix sign of temperature measurements (Closes: #29) (#30)