syssi/esphome-basen-bms #32: Doesn't work on 51.2V 200ah LiFePO4

yutani42, 2024-03-17:
Hi,

might be obvious (?) but I tried Your solution on a BasenGreen 10KW Wall-Mounted 51.2V 200ah LiFePO4
https://www.basengreenshop.com/48v-lfp-series/basengreen-10kw-wall-mounted-51-2v-200ah-lifepo4-battery-pack-for-solar-energy-storage
and it doesn't seem to work:

[image]

Any chance this could work somehow?

syssi, 2024-03-17:
Whats the advertised bluetooth name of the BMS?

yutani42, 2024-03-17:
TP_BSTBD-23I-401

yutani42, 2024-03-18:
Now realizing it might be a tianpower BMS i tried Your other repo, but to no avail:
[image]

syssi, 2024-03-18:
It looks like you have missed to update the `mac_address` substitution to the MAC address of your device.

yutani42, 2024-03-18:
Oh boy, I missed that, now it works, I almost can't believe it!

Thanks for Your help and great work!

syssi, 2024-03-18:
Could you provide some details of your battery? I would like to extend the README of the tianpower BMS project.

yutani42, 2024-03-18:
Sure, what do You want to know (the store page pretty much gives all the tech-details?)

syssi, 2024-03-18:
This is my paypal handle: paypal.me/syssi84