syssi/esphome-basen-bms #36: Basen Green 48V DIY KITS V2

straga, 2024-08-07:
Anybody have information about protocol or some else.

[BASENGREEN-48V-DIY-KITS_User_Manual.pdf](https://github.com/user-attachments/files/16519075/BASENGREEN-48V-DIY-KITS_User_Manual.1.pdf)

What inside:

[image]
[image]
[image]
[image]
[image]

syssi, 2024-08-07:
Does the BMS support bluetooth? What's the advertised bluetooth name?

syssi, 2024-08-07:
I could be a Tianpower BMS if the bluetooth name starts with `TP`: https://github.com/syssi/esphome-tianpower-bms

[image]

syssi, 2024-08-07:
Could you start the Windows application and talk to your BMS? You can configure the protocol of the CAN bus and RS485 bus here:

[image]

Could you list all available protocols?